OpenVADL/openvadl #154: ast: Crashes with type error but unclear why

kper, 2025-04-11:
The following specification

```

instruction set architecture Tests = {
  using Index = Bits<5>
  using Address  = Bits<32>

  register file X: Index -> Bits<32>

  instruction SHIFT_LEFT_1: TMP = {
    X(ONE) := (X(ONE) << 1) as Bits<32>
  }

  encoding SHIFT_LEFT_1 = { ONE = 2 }

  assembly SHIFT_LEFT_1 = "just a dummy"

  format TMP: Address =
  { ONE: Bits<5>
  }
}

micro processor TEST implements Tests = { start = 0x1000 }
```

crashes with the error message

```
    error: Invalid Format
         ╭──[../../../../../tmp/OpenVADL-testSource-5395563716916932919/passes/algebraicSimplification/algebraic_simplification.vadl:16:3]
         │
      16 │>   format TMP: Address =
      17 │>   { ONE: Bits<5>
      18 │>   }
         │    
         │ 
         Declared as bitwidth of size 32 but actually size is 5
```

Isn't the bitwidth still 5?

flofriday, 2025-04-11:
We can definitley improve the error message but you declared the format `TMP` with a size of 32 Bits but all fields added up are only 5 Bits.

But I also do not know if this _should_ be an error, I had to guess a lot about the semantics here and maybe I got that wrong.

kper, 2025-04-11:
ah ok

the `format TMP: Address =` was the problem

thanks

closed 2025-04-11 by kper