OpenVADL/openvadl #163: frontend: Crash when using ISA definition processor

Jozott00, 2025-04-14:
The following code
```
instruction set architecture ISA = {
  memory MEM: Bits<64> -> Bits<8>
}

micro processor CPU implements ISA = {
  start = 0x1000
  firmware = {
    MEM(0x1000) := 0x1
   }
}
```

crashes with
```
...
java.lang.IllegalStateException: Cannot find symbol `MEM` found at: file:///Users/jozott/development/uni/dipl/iss-dev/code/open-vadl/sys/risc-v/other.vadl:8:5
	at vadl.ast.TypeChecker.visitIdentifiable(TypeChecker.java:1868)
	at vadl.ast.TypeChecker.visit(TypeChecker.java:1873)
	at vadl.ast.TypeChecker.visit(TypeChecker.java:66)
	at vadl.ast.Identifier.accept(Expr.java:168)
...
```

Also when changing `MEM(0x1000)` to `MEM<4>(0x1000)` it crashes with an other error:
```
java.lang.IllegalStateException
	at vadl.ast.TypeChecker.visit(TypeChecker.java:2802)
	at vadl.ast.TypeChecker.visit(TypeChecker.java:66)
	at vadl.ast.CallIndexExpr.accept(Expr.java:1624)
	at vadl.ast.TypeChecker.check(TypeChecker.java:124)
	at vadl.ast.TypeChecker.visit(TypeChecker.java:3076)
```

This is strange, as the [RV64IM processor specification](https://github.com/OpenVADL/openvadl/blob/04da9882e8fec8307f5c1dfef29aeb523d105c48/vadl/test/resources/testSource/sys/risc-v/rv64im.vadl#L122-L143) works fine.

flofriday, 2025-04-16:
Because so far definitions behave like "scopes", which means that variables from a ISA aren't visible to microarchitectures.

I think you could fix that by fully qualifying the variables:

```
instruction set architecture ISA = {
  memory MEM: Bits<64> -> Bits<8>
}

micro processor CPU implements ISA = {
  start = 0x1000
  firmware = {
    ISA::MEM(0x1000) := 0x1
   }
}
```

But for 1) obviously this shouldn't have crashed but thrown an error and 2) I guess this is common enough that always all variables from the implemented isa should be in scope but 3) I guess this means that all global symbols and all symbols from the implemented isa should be in scope which means there is a big refactoring waiting for me because for now scopes could only have a single enclosing scope.

flofriday, 2025-04-21:
Now that I have rewritten the symbol resolver in #180 I can tell you that sharing "scopes" isn't that rare and we already to it in a couple of places, I guess we just also need to add it to the microprocesssor.

But at least it now doesn't crash anymore 👍

Jozott00, 2025-05-03:
I found the issue:
The symbol table of the `processor` definition was not extended by the symbol table of the `ISA`s it implements.
```
          definition.symbolTable().extendBy(isa.symbolTable());
```

Fixed in #212.

closed 2025-05-03 by Jozott00