OpenVADL/openvadl #217: frontend: crashes with the aarch64 spec

AndreasKrall, 2025-05-04:
The parser crashes with the new open source `aarch64.vadl` spec in `sys/aarch64/`.
If the last model invocation is uncommented, no models are called and no crash happens:
```
// base instructions ***********************************************************

  $InstrWXFlags     (AddSubImmInstr           ; (ADD    ; "add"   ; 0b0'1000'10      ; adds  ))

```
```
                             ___ ___    _   ___ _  _
                            / __| _ \  /_\ / __| || |
                           | (__|   / / _ \\__ \ __ |
                            \___|_|_\/_/ \_\___/_||_|

                       🔥 The OpenVADL compiler crashed 🔥

This shouldn't have happened, please open an issue with the stacktrace below at:
https://github.com/OpenVADL/open-vadl/issues/new

java.lang.IndexOutOfBoundsException: Index 3 out of bounds for length 3
	at java.base@21.0.4/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:100)
	at java.base@21.0.4/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:106)
	at java.base@21.0.4/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:302)
	at java.base@21.0.4/java.util.Objects.checkIndex(Objects.java:385)
	at java.base@21.0.4/java.util.ArrayList.get(ArrayList.java:427)
	at vadl.ast.MacroExpander.resolveArg(MacroExpander.java:1184)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:225)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:42)
	at vadl.ast.PlaceholderExpr.accept(Expr.java:775)
	at vadl.ast.MacroExpander.expandExpr(MacroExpander.java:69)
	at vadl.ast.MacroExpander.expandNode(MacroExpander.java:157)
	at java.base@21.0.4/java.util.ArrayList.replaceAllRange(ArrayList.java:1795)
	at java.base@21.0.4/java.util.ArrayList.replaceAll(ArrayList.java:1785)
	at vadl.ast.MacroExpander.expandNode(MacroExpander.java:164)
	at vadl.ast.MacroExpander.collectMacroParameters(MacroExpander.java:1075)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:659)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:42)
	at vadl.ast.MacroInstanceDefinition.accept(Definition.java:2527)
	at vadl.ast.MacroExpander.expandDefinition(MacroExpander.java:114)
	at vadl.ast.MacroExpander.expandDefinitions(MacroExpander.java:103)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:682)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:42)
	at vadl.ast.DefinitionList.accept(Definition.java:2857)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:662)
	at vadl.ast.MacroExpander.visit(MacroExpander.java:42)
	at vadl.ast.MacroInstanceDefinition.accept(Definition.java:2527)
	at vadl.ast.MacroExpander.expandDefinition(MacroExpander.java:114)
	at vadl.ast.MacroExpander.expandNode(MacroExpander.java:159)
	at vadl.ast.ParserUtils.expandNode(ParserUtils.java:541)
	at vadl.ast.Parser.macroReplacement(Parser.java:678)
	at vadl.ast.Parser.isaDefinitionList(Parser.java:849)
	at vadl.ast.Parser.instructionsetDefinition(Parser.java:437)
	at vadl.ast.Parser.topLevelDefinitions(Parser.java:339)
	at vadl.ast.Parser.vadl(Parser.java:312)
	at vadl.ast.Parser.Parse(Parser.java:3791)
	at vadl.ast.VadlParser.parse(VadlParser.java:115)
	at vadl.ast.VadlParser.parse(VadlParser.java:62)
	at vadl.cli.BaseCommand.parseToAst(BaseCommand.java:140)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:221)
	at vadl.cli.BaseCommand.call(BaseCommand.java:298)
	at vadl.cli.BaseCommand.call(BaseCommand.java:61)
	at picocli.CommandLine.executeUserObject(CommandLine.java:2045)
	at picocli.CommandLine.access$1500(CommandLine.java:148)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2465)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2457)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2419)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:2277)
	at picocli.CommandLine$RunLast.execute(CommandLine.java:2421)
	at picocli.CommandLine.execute(CommandLine.java:2174)
	at vadl.cli.Main.main(Main.java:44)
	at java.base@21.0.4/java.lang.invoke.LambdaForm$DMH/sa346b79c.invokeStaticInit(LambdaForm$DMH)
```

AndreasKrall, 2025-05-06:
And another crash with wrong macros:
```
                             ___ ___    _   ___ _  _
                            / __| _ \  /_\ / __| || |
                           | (__|   / / _ \\__ \ __ |
                            \___|_|_\/_/ \_\___/_||_|

                       🔥 The OpenVADL compiler crashed 🔥

This shouldn't have happened, please open an issue with the stacktrace below at:
https://github.com/OpenVADL/open-vadl/issues/new

java.util.NoSuchElementException
	at java.base@21.0.4/java.util.ArrayList$Itr.next(ArrayList.java:1052)
	at vadl.ast.ParserUtils$1.next(ParserUtils.java:328)
	at vadl.ast.ParserUtils$1.next(ParserUtils.java:318)
	at vadl.ast.Parser.maximumMacroExpr(Parser.java:3551)
	at vadl.ast.Parser.macroReplacement(Parser.java:672)
	at vadl.ast.Parser.isaDefinitionList(Parser.java:849)
	at vadl.ast.Parser.macroBody(Parser.java:3688)
	at vadl.ast.Parser.macroDef(Parser.java:398)
	at vadl.ast.Parser.isaDefinitionList(Parser.java:846)
	at vadl.ast.Parser.instructionsetDefinition(Parser.java:437)
	at vadl.ast.Parser.topLevelDefinitions(Parser.java:339)
	at vadl.ast.Parser.vadl(Parser.java:312)
	at vadl.ast.Parser.Parse(Parser.java:3791)
	at vadl.ast.VadlParser.parse(VadlParser.java:115)
	at vadl.ast.VadlParser.parse(VadlParser.java:62)
	at vadl.cli.BaseCommand.parseToAst(BaseCommand.java:140)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:221)
	at vadl.cli.BaseCommand.call(BaseCommand.java:298)
	at vadl.cli.BaseCommand.call(BaseCommand.java:61)
	at picocli.CommandLine.executeUserObject(CommandLine.java:2045)
	at picocli.CommandLine.access$1500(CommandLine.java:148)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2465)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2457)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2419)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:2277)
	at picocli.CommandLine$RunLast.execute(CommandLine.java:2421)
	at picocli.CommandLine.execute(CommandLine.java:2174)
	at vadl.cli.Main.main(Main.java:44)
	at java.base@21.0.4/java.lang.invoke.LambdaForm$DMH/sa346b79c.invokeStaticInit(LambdaForm$DMH)
```

flofriday, 2025-05-15:
I investigated this and the implementation did crash when too many arguments were provided, something my PR #251  adresses.

closed 2025-05-15 by Jozott00