OpenVADL/openvadl #227: macro-system: Make `with` construct more powerful

Jozott00, 2025-05-06:
In VADL one can overwrite the implementation of a macro using `with` in the `import` declaration or via command line. 

However, this is limited to expression macros, and the `with` definition is a String. This makes is very limited in its capabilities. 

Let's consider this to be `a.vadl`
```
instruction set architecture TEST = {
  model size() : Val = { 64 }
  register X: Bits<$size()>
}
```
I can do something like this in `b.vadl` (for some reason that does not work at the moment, but I think it should). 
```
import a::TEST with ("size=64")
```

However, what is not possible is something like this:
```
model mysize() : Val = { 64 }
import a::TEST with ("size=$mysize()")
```

Additionally, only `MacroInstanceExpr` can be overwritten, which makes it not possible to overwrite things like definitions. 

In my understanding, a generalized support for such macro overwrites should not be too hard. Instead of a String, we would define it using 
```
model mysize() : Val = { 64 }
import a::TEST with (size=$mysize())
```

## Instruction Overwriting

For me, the real reason for the necessity of this function is the RISC-V commands ECALL and EBREAK. In the base ISA they are only NOPs, whereas in the privileged ISA they are raising exceptions.

As explained in #40, multiple inheritance would be a good way to represent this, but at the moment we don't have a precise way to specify this.

If we had the feature described above, we could do the following:
In `rv3264im.vadl` we define the `ECALL` and `EBREAK` with
```
  model ECallInstr (name : Id, imm : Lit, impl : Stats) : IsaDefs = {
    instruction $name : Itype = { $impl }                    // ecall / ebreak instructions
    encoding $name = {opcode = 0b111'0011, funct3 = 0b000, rd = 0b0'0000, rs1 = 0b0'0000, imm = $imm}
    assembly $name = (mnemonic)
    }

  model EBreakCall() : IsaDefs = {
      $ECallInstr (ECALL ; 0 ; {} )                              // environment (sys) call
      $ECallInstr (EBREAK; 1 ; {} )                              // environment (sys) break
  }

  $EBreakCall()
```

This would make it possible to override the implementation in the privileged RISC-V ISA (that extends the base ISA) with
```
model NoEBreakCall : IsaDefs = {}
import rv3264im::RV3264IM with ( EBreakCall = $NoEBreakCall() )

instruction set architecture Priv extending RV3264IM {
  // Privileged implementation of EBREAK and ECALL
  ...
}
```

AndreasKrall, 2025-05-07:
> I can do something like this in `b.vadl` (for some reason this does not work at the moment, but I think it should).
> 
> ```
> import a::TEST with ("size=64")
> ```

Currently only `Id` is supported, the extension for `Int` or even expressions is planned.

AndreasKrall, 2025-05-07:
Related is #105. With having all top-level names placeholders, it would be easier to group specifications.

Currently `with` and the command line have the same expressiveness. With this planned extension eventually it could be difficult to support them on the same level and we have to think about string escape solutions in the `with` string compatible for command line and `with`.

I agree, that multiple ISA inheritance would be best.