OpenVADL/openvadl #230: parser: Crash in parser on macro definition

Jozott00, 2025-05-06:
The input program 
```
model m(y: Val, x: Ex) : Stmt = {
}
```
crashes with
```
java.lang.IllegalStateException: Internal error in grammar at token (2:24): Unhandled syntax type INVALID
	at vadl.ast.Parser.internalError(Parser.java:3805)
	at vadl.ast.Parser.macroBody(Parser.java:3796)
	at vadl.ast.Parser.macroDef(Parser.java:400)
	at vadl.ast.Parser.vadl(Parser.java:323)
	at vadl.ast.Parser.Parse(Parser.java:3814)
	at vadl.ast.VadlParser.parse(VadlParser.java:115)
	at vadl.ast.VadlParser.parse(VadlParser.java:62)
	at vadl.cli.BaseCommand.parseToAst(BaseCommand.java:140)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:221)
	at vadl.cli.BaseCommand.call(BaseCommand.java:298)
	at vadl.cli.BaseCommand.call(BaseCommand.java:61)
```

Jozott00, 2025-05-06:
The reason is that `Stmt` is not a korrekt syntax type. The correct syntax type would be `Stat`.

This is still a bug, as it should not crash, but explain the error.

AndreasKrall, 2025-05-07:
> The input program
>
> ```
> model m(y: Val, x: Ex) : Stmt = {
> }
> ```
>
> crashes with
>
> ```
> java.lang.IllegalStateException: Internal error in grammar at token (2:24): Unhandled syntax type INVALID
> 	at [...]

This error is related to #211, but gives a different stack trace.

closed 2025-05-07 by Jozott00