OpenVADL/openvadl #240: frontend: `smull`, `umull` the type checker does no recognize the double sized result

AndreasKrall, 2025-05-09:
The following test raises an syntax error:
```
instruction set architecture Test = {
  register file X : Bits<5> -> Bits<64>  // general purpose register file with stack pointer
  format ThreeRegOpFormat: Bits<32> =    // three register operand format
    { sf       [31]                      // size field, if (sf = 0) 32 bit operation else 64 bit
    , op       [30..21,15..10]           // opcode
    , rm       [20..16]                  // 2nd source register
    , rn       [9..5]                    // 1st source register
    , rd       [4..0]                    // destination register
    }

  instruction mulh : ThreeRegOpFormat =
    let result = VADL::smull (X(rn), X(rm)) in
      X(rd) := result(127..64)
}
```
```
error: Invalid Range
     ╭──[test.vadl:13:23]
     │
  13 │       X(rd) := result(127..64)
     │                       ^^^^^^^ 
     │ 
     Range start 127 out of bounds for `Bits<64>`
```
If instead of `VADL::smull` the operator version `*#` is used, everything works fine.

Jozott00, 2025-05-09:
I noticed this too, I think the operator `*#` correctly produces a double sized output type, but the built-in does not.

AndreasKrall, 2025-06-01:
It works now, but a test like the following would be fine:
```
instruction set architecture Test = {
  register X : Bits<5> -> Bits<64>       // general purpose register file with stack pointer
  format ThreeRegOpFormat: Bits<32> =    // three register operand format
    { sf       [31]                      // size field, if (sf = 0) 32 bit operation else 64 bit
    , op       [30..21,15..10]           // opcode
    , rm       [20..16]                  // 2nd source register
    , rn       [9..5]                    // 1st source register
    , rd       [4..0]                    // destination register
    }

  model ThreeRegOpEncAsm (i: Id, op: Lit): IsaDefs = {
    encoding $i = { op = $op, sf = 1 }
    assembly $i = (mnemonic, ' ', register(rd), ', ', register(rn), ', ', register(rm))
    }

  model ThreeRegOp (i: Id, op: Lit): IsaDefs = {
    instruction $i : ThreeRegOpFormat =
      let result = VADL::$i (X(rn), X(rm)) in
        X(rd) := result(127..64)
    $ThreeRegOpEncAsm ($i; $op)
    }

  model ThreeRegOpFlags (i: Id, op: Lit): IsaDefs = {
    instruction $i : ThreeRegOpFormat =
      let result, flags = VADL::$i (X(rn), X(rm)) in
        X(rd) := result(127..64)
    $ThreeRegOpEncAsm ($i; $op)
    }

  $ThreeRegOp (smull;  0)
  $ThreeRegOp (umull;  1)
  $ThreeRegOp (sumull; 2)
  $ThreeRegOpFlags (smulls;  10)
  $ThreeRegOpFlags (umulls;  11)
  $ThreeRegOpFlags (sumulls; 12)
}
```

closed 2025-06-06 by Jozott00