OpenVADL/openvadl #244: frontend: crashes with ppc specification

kbschlosser, 2025-05-10:
```
                             ___ ___    _   ___ _  _
                            / __| _ \  /_\ / __| || |
                           | (__|   / / _ \\__ \ __ |
                            \___|_|_\/_/ \_\___/_||_|

                       ? The OpenVADL compiler crashed ?

This shouldn't have happened, please open an issue with the stacktrace below at:
https://github.com/OpenVADL/open-vadl/issues/new

java.util.NoSuchElementException
	at java.base/java.util.ArrayList.getFirst(ArrayList.java:439)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:977)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:44)
	at vadl.ast.BehaviorLowering.buildBranch(BehaviorLowering.java:348)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1121)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:42)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1076)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:38)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1165)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:40)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1076)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:38)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1165)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:40)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1076)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:38)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1165)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:96)
	at vadl.ast.Statement.accept(Statement.java:40)
	at vadl.ast.BehaviorLowering.getInstructionGraph(BehaviorLowering.java:154)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:1104)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:102)
	at vadl.ast.InstructionDefinition.accept(Definition.java:1359)
	at vadl.ast.ViamLowering.fetch(ViamLowering.java:164)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
	at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
	at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:622)
	at java.base/java.util.stream.ReferencePipeline.toList(ReferencePipeline.java:627)
	at vadl.ast.ViamLowering.visitIsa(ViamLowering.java:1126)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:1218)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:102)
	at vadl.ast.ProcessorDefinition.accept(Definition.java:3839)
	at vadl.ast.ViamLowering.fetch(ViamLowering.java:164)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at vadl.ast.ViamLowering.generate(ViamLowering.java:141)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:234)
	at vadl.cli.BaseCommand.call(BaseCommand.java:298)
	at vadl.cli.BaseCommand.call(BaseCommand.java:61)
	at picocli.CommandLine.executeUserObject(CommandLine.java:2045)
	at picocli.CommandLine.access$1500(CommandLine.java:148)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2465)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2457)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2419)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:2277)
	at picocli.CommandLine$RunLast.execute(CommandLine.java:2421)
	at picocli.CommandLine.execute(CommandLine.java:2174)
	at vadl.cli.Main.main(Main.java:44)
```
attached as txt file, because vadl files are not supported for upload.
[ppc.txt](https://github.com/user-attachments/files/20140701/ppc.txt)

Jozott00, 2025-05-11:
@kbschlosser can you provide more information on what specification change led to the crash?

kbschlosser, 2025-05-11:
@Jozott00 The only major change is due to the transition from register file to register.
```
  register file       GPR  : GPRIndex -> Reg
  [GPRZ(0) = 0]
  alias register file GPRZ : GPRIndex -> Reg = GPR
```
->
```
  register GPR  : GPRIndex -> Reg
  //[GPRZ(0) = 0]
  //TODO: should be "alias register GPRZ : GPRIndex -> Reg = GPR"
  register GPRZ : GPRIndex -> Reg
```
After pulling the newest version of openvadl from git recently, I received new errors relating to 'file'. After making the mentioned change, I now get the error in the issue above.