OpenVADL/openvadl #263: frontend: Error in type checking of vector types

AndreasKrall, 2025-05-18:
When checking the following test program
```
instruction set architecture Test = {
  using Byte = Bits<8>
  using Byte8 = Byte<8>

  register X : Bits<5> -> Bits<64>
  
  format testFormat : Byte = {op: Byte}

  instruction test : testFormat =
    let xByte8 = X(1) as Byte8 in
      X(31) := xByte8
  }
```
The type checker wrongly gives a type mismatch error:
```
error: Type Mismatch
     ╭──[test.vadl:11:16]
     │
  11 │       X(31) := xByte8
     │                ^^^^^^ Expected `Bits<64>` but got `Bits<8>`.
     │ 
```
xByte is a 64 bit type not an 8 bit type.

flofriday, 2025-06-08:
Agreed the typechecking for tensors is quite wrong (currently working on a fix).
But I would still argue that we shouldn't allow implicit casting from vector types with different dimensions as it can hide quite a bit of complexity and instead reject the program with the following error:

```
error: Type Mismatch
     ╭──[test.vadl:11:16]
     │
  11 │       X(31) := xByte8
     │                ^^^^^^ Expected `Bits<64>` but got `Bits<8><8>`.
     │ 
```

And require an explicit cast here: `X(31) := xByte8 as Bits<64>`

AndreasKrall, 2025-06-09:
I am fine with an explicit cast, but nevertheless we should discuss this in our next meeting. Please implement it currently with an explicit cast here.

closed 2025-06-16 by Jozott00