OpenVADL/openvadl #304: ast: Symbol name already used varrisc

kper, 2025-05-28:
I tried to create a lcb from the varrisc specification and copied the `rv32im.vadl` abi.
However, I get an error from the typechecker.

```
error: Symbol name already used: offset
     ╭──[sys/v-risc/VarRisc.vadl:398:42]
     │
 395 │   pseudo instruction BGEZ( rs1: Index, offset : SInt<11> ) = {
     │                                        ------ First defined here.
     ⋮
 398 │   pseudo instruction BGEZ_L( rs1: Index, offset : SIntR ) = {
     │                                          ^^^^^^ Second definition here.
     │
     note: All symbols must have a unique name.
```

```
// SPDX-FileCopyrightText : © 2024 TU Wien <vadl@tuwien.ac.at>
// SPDX-License-Identifier: Apache-2.0

import VarRisc::{VarRISC32}

application binary interface ABI for VarRISC32 = {
   size_t type = unsigned int
   pointer align = 32

   //
   // Register alias definitions
   //

   alias register zero = X(0)

   alias register ra = X(1)

   alias register sp = X(2)

   alias register gp = X(3)
   alias register tp = X(4)

   alias register t0 = X(5)
   alias register t1 = X(6)
   alias register t2 = X(7)

   // [ preferred alias ]
   alias register fp = X(8)

   alias register s0 = X(8)
   alias register s1 = X(9)

   alias register a0 = X(10)
   alias register a1 = X(11)
   alias register a2 = X(12)
   alias register a3 = X(13)
   alias register a4 = X(14)
   alias register a5 = X(15)
   alias register a6 = X(16)
   alias register a7 = X(17)

   alias register s2 = X(18)
   alias register s3 = X(19)
   alias register s4 = X(20)
   alias register s5 = X(21)
   alias register s6 = X(22)
   alias register s7 = X(23)
   alias register s8 = X(24)
   alias register s9 = X(25)
   alias register s10 = X(26)
   alias register s11 = X(27)

   alias register t3 = X(28)
   alias register t4 = X(29)
   alias register t5 = X(30)
   alias register t6 = X(31)

   // [ alignment : HalfWord ]
   stack pointer = sp
   return address = ra
   global pointer = gp
   frame pointer = fp
   thread pointer = tp

   special return instruction = RET
   special call instruction = CALL
   special local address load instruction = LLA
   special global address load instruction = LGA_32
   special absolute address load instruction = LA

   return value = a{0..1}
   function argument = a{0..7}

   // ra is callee saved because it is used as normal register and has to be restored.
   caller saved = [ a{0..7}, t{0..6} ]
   callee saved = [ ra, gp, tp, fp, s{0..11} ]

  constant sequence( rd : Bits<5>, val : SInt<32> ) =
  {
     LUI { rd = rd, imm = hi( val ) }
     ADDI { rd = rd, rs1 = rd, imm = lo( val ) }
  }

  constant sequence( rd : Bits<5>, val : UInt<32> ) =
  {
     LUI { rd = rd, imm = hi( val ) }
     ADDI { rd = rd, rs1 = rd, imm = lo( val ) }
  }

  constant sequence( rd : Bits<5>, imm : SInt<12> ) =
  {
     ADDI{ rd = rd, rs1 = 0, imm = imm }
  }

  register adjustment sequence( rd : Bits<5>, rs1: Bits<5>, imm : SInt<12> ) =
  {
     ADDI{ rd = rd, rs1 = rs1, imm = imm }
  }
}

[ comment string: "#" ]
assembly description ASM for ABI = {

  modifiers = {
    "hi" -> RV32IM::hi,
    "lo" -> RV32IM::lo
  }

  grammar = {
    RRIds : "ADD" | "SUB" | "AND" | "OR" | "XOR" | "SLT" | "SLTU" | "SLL" | "SRL" | "SRA"
            | "MUL" | "MULH" | "MULHSU" | "MULHU" | "DIV" | "DIVU" | "REM" | "REMU"
    ;

    RRInstruction @instruction:
      mnemonic = RRIds @operand
      rd = Register@operand ","
      rs1 = Register@operand ","
      rs2 = Register@operand
    ;

    ITypeIds : "ANDI" | "ORI" | "XORI" | "SLTI" | "SLTIU" | "JALR" ;

    ITypeInstruction @instruction:
      mnemonic = ITypeIds @operand
      rd = Register@operand ","
      rs1 = Register@operand ","
      imm = ImmediateOperand
    ;

    AddIInstruction @instruction:
      mnemonic = "ADDI" @operand
      rd = Register@operand ","
      rs1 = Register@operand ","
      imm = LoImmediateOperand
    ;

    LoImmediateOperand :
      ImmediateOperand
      | op = (
          "%" mod="lo"@modifier
          "(" val = Expression ")"
        ) @operand
    ;

    LuiInstruction @instruction:
      mnemonic = "LUI" @operand
      rd = Register@operand ","
      imm = LuiImmediateOperand
    ;

    LuiImmediateOperand :
      ImmediateOperand
      | op = (
          "%" mod="hi"@modifier
          "(" val = Expression ")"
        ) @operand
    ;

    AuipcInstruction @instruction:
      mnemonic = "AUIPC" @operand
      rd = Register@operand ","
      imm = ImmediateOperand
    ;

    LTypeIds : "LB" | "LBU" | "LH" | "LHU" | "LW" ;

    LTypeInstruction @instruction:
      mnemonic = LTypeIds @operand
      rd = Register @operand ","
      imm = LoImmediateOperand
      "(" rs1 = Register @operand ")"
    ;

    STypeIds : "SB" | "SH" | "SW" ;

    STypeInstruction @instruction:
      mnemonic = STypeIds @operand
      rs2 = Register @operand ","
      imm = LoImmediateOperand
      "(" rs1 = Register @operand ")"
    ;

    JalInstruction @instruction:
      mnemonic = "JAL" @operand
      rd = Register @operand ","
      immS = ImmediateOperand
    ;

    BTypeIds : "BEQ" | "BNE" |  "BGE" | "BGEU" | "BLT" | "BLTU" ;

    BTypeInstruction @instruction:
      mnemonic = BTypeIds @operand
      rs1 = Register@operand ","
      rs2 = Register@operand ","
      immS = ImmediateOperand@operand
    ;

    IShftTypeIds : "SLLI" | "SRLI" | "SRAI" ;

    IShftTypeInstruction @instruction:
      mnemonic = IShftTypeIds @operand
      rd = Register@operand ","
      rs1 = Register@operand ","
      sft = ImmediateOperand
    ;

    NoOperandIds : "RET" | "NOP" | "ECALL" | "EBREAK" ;

    NoOperandPseudoInstruction @instruction :
      mnemonic = NoOperandIds @operand
    ;

    SymbolPseudoIds : "CALL" | "TAIL" ;

    SymbolPseudoInstruction @instruction :
      mnemonic = SymbolPseudoIds @operand
      symbol = (Identifier @symbol) @operand
    ;

    JPseudoInstruction @instruction :
      mnemonic = "J" @operand
      offset = ImmediateOperand
    ;

    RRPseudoIds : "MV" | "NOT" | "NEG" | "SNEZ" | "SLTZ" | "SGTZ" ;

    RRPseudoInstruction @instruction :
      mnemonic = RRPseudoIds @operand
      rd = Register @operand ","
      rs1 = Register @operand
    ;

    BranchPseudoIds : "BEQZ" | "BNEZ" | "BLEZ" | "BGEZ" | "BLTZ" | "BGTZ" ;

    BranchPseudoInstruction @instruction :
      mnemonic = BranchPseudoIds @operand
      rs = Register @operand ","
      offset = ImmediateOperand
    ;

    LAPseudoInstruction @instruction :
      mnemonic = "LA" @operand
      rd = Register @operand ","
      symbol = ImmediateOperand
    ;

    LLAPseudoInstruction @instruction:
      mnemonic = "LLA" @operand
      rd = Register @operand ","
      symbol = ImmediateOperand
    ;

    LIPseudoInstruction @instruction:
      mnemonic = "LI" @operand
      rd = Register @operand ","
      symbol = ImmediateOperand
    ;
  }
}

[ htif ]
processor Spike implements RV32IM with ABI = {
  reset = PC := 0x80000000

  [ firmware ]
  [ base: 0x80000000 ]
  memory region [RAM] DRAM in MEM

}
```

AndreasKrall, 2025-05-29:
The ABI file in this issue is for RISC-V and cannot work for VarRisc. Nevertheless, the error also happens without this file. The problem lies in the assembly specification for multiple instructions. It does not occur, if the assembly specs are separated. But when the assembly specs are separated, it triggers a crash, probably related to the pseudo instruction calling another pseudo instruction:
```
                             ___ ___    _   ___ _  _
                            / __| _ \  /_\ / __| || |
                           | (__|   / / _ \\__ \ __ |
                            \___|_|_\/_/ \_\___/_||_|

                       🔥 The OpenVADL compiler crashed 🔥

This shouldn't have happened, please open an issue with the stacktrace below at:
https://github.com/OpenVADL/open-vadl/issues/new

java.lang.IllegalStateException: The behavior generator doesn't implement yet
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1207)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:103)
	at vadl.ast.Statement.accept(Statement.java:50)
	at vadl.ast.BehaviorLowering.lambda$getInstructionSequenceGraph$0(BehaviorLowering.java:190)
	at java.base@21.0.4/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base@21.0.4/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:622)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toList(ReferencePipeline.java:627)
	at vadl.ast.BehaviorLowering.getInstructionSequenceGraph(BehaviorLowering.java:191)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:1398)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:108)
	at vadl.ast.PseudoInstructionDefinition.accept(Definition.java:1485)
	at vadl.ast.ViamLowering.fetch(ViamLowering.java:225)
	at java.base@21.0.4/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base@21.0.4/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:622)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toList(ReferencePipeline.java:627)
	at vadl.ast.ViamLowering.visitIsa(ViamLowering.java:1189)
	at vadl.ast.ViamLowering.findLeafIsa(ViamLowering.java:196)
	at vadl.ast.ViamLowering.generate(ViamLowering.java:153)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:241)
	at vadl.cli.BaseCommand.call(BaseCommand.java:305)
	at vadl.cli.BaseCommand.call(BaseCommand.java:62)
	at picocli.CommandLine.executeUserObject(CommandLine.java:2045)
	at picocli.CommandLine.access$1500(CommandLine.java:148)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2465)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2457)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2419)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:2277)
	at picocli.CommandLine$RunLast.execute(CommandLine.java:2421)
	at picocli.CommandLine.execute(CommandLine.java:2174)
	at vadl.cli.Main.main(Main.java:44)
	at java.base@21.0.4/java.lang.invoke.LambdaForm$DMH/sa346b79c.invokeStaticInit(LambdaForm$DMH)
```

kper, 2025-05-29:
@AndreasKrall but this is a different error. Can you create an extra issue for that?
I have edited the issue so that the problem with the _`offset` symbol is used multiple times_ is covered.

flofriday, 2025-05-29:
@AndreasKrall the issue here arises because in the assembly `offset` refers to two different parameters (from the two different pseudo instructions). 
The syntax tables current cannot deal with that, which is fixable, but my problem is that the syntax here can be difficult to understand.

Consider the following:
```
  pseudo instruction A( offset : SInt<8> ) = {
    ...
  }
  pseudo instruction B( offset : SInt<9> ) = {
    ...
  }
  assembly A, B = decimal(offset + (1 as SInt<8>)) 
```

This would throw an error, even though when typechecked with the scope from `A` everything is correct.

I think only allowing one instruction per assembly would make debugging a lot easier.
Also I looked over the sourcecode and we only used the multiple instruction assembly in testcases but in none of the proper specs.