OpenVADL/openvadl #311: sys/aarch64.vadl: Missing instruction `BTI` in AArch64 spec

Jozott00, 2025-05-28:
Five Embench benchmarks utilize the instruction.  This instruction should decode as a NOP instruction in our implementation.

https://developer.arm.com/documentation/ddi0602/2025-03/Base-Instructions/BTI--Branch-target-identification-?lang=en

AndreasKrall, 2025-05-28:
Are implemented now, please check them if they are working?

Jozott00, 2025-05-29:
In the current spec, the `BTI` instruction has the fixed op2 `00`, however in the test is uses `bti c` (so the op2 is `01`). 

It can be implemented as 
```
    function btiOp2(op2: Bits<2>) -> String = match op2 with
    { 0b01 => " c"
    , 0b10 => " j"
    , 0b11 => " jc"
    , _    => ""
    }

    format BTIFormat: Instr =
    { op      [31..8,5..0]
    , op2     [7..6]
    }
    instruction BTI: BTIFormat = { }
    encoding BTI = { op = 0b110101010000001100100100011111 }
    assembly BTI = ( "bti", btiOp2(op2) )
```

However, I guess you want it to look a bit different in the `aarch64.vadl`.

AndreasKrall, 2025-05-29:
I have fixed that and added rev16 too. Please check.