OpenVADL/openvadl #312: frontend: Crash (probably when a pseudo instruction calls another pseudo instruction)

AndreasKrall, 2025-05-29:
The following code from VarRisc:
```
  pseudo instruction BRA( offset : SInt<11> ) = {
    BEQZ( 0, offset )
  }
  pseudo instruction BRA_L( offset : SIntR ) = {
    BEQZ_L( 0, offset )
  }
  assembly BRA   = ( mnemonic, ' ', decimal(offset) )
  assembly BRA_L = ( mnemonic, ' ', decimal(offset) )

  pseudo instruction BEQZ( rd: Index, offset : SInt<11> ) = {
    BEQ{ rd = rd, rs1 = 0, imm = offset }
  }
  pseudo instruction BEQZ_L( rd: Index, offset : SIntR ) = {
    BEQ_L{ rd = rd, rs1 = 0, imm = offset }
  }
  assembly BEQZ   = ( mnemonic, ' ', register(rd), ',', decimal(offset) )
  assembly BEQZ_L = ( mnemonic, ' ', register(rd), ',', decimal(offset) )
```
gives the following crash:
```
                             ___ ___    _   ___ _  _
                            / __| _ \  /_\ / __| || |
                           | (__|   / / _ \\__ \ __ |
                            \___|_|_\/_/ \_\___/_||_|

                       🔥 The OpenVADL compiler crashed 🔥

This shouldn't have happened, please open an issue with the stacktrace below at:
https://github.com/OpenVADL/open-vadl/issues/new

java.lang.IllegalStateException: The behavior generator doesn't implement yet
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:1207)
	at vadl.ast.BehaviorLowering.visit(BehaviorLowering.java:103)
	at vadl.ast.Statement.accept(Statement.java:50)
	at vadl.ast.BehaviorLowering.lambda$getInstructionSequenceGraph$0(BehaviorLowering.java:190)
	at java.base@21.0.4/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base@21.0.4/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:622)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toList(ReferencePipeline.java:627)
	at vadl.ast.BehaviorLowering.getInstructionSequenceGraph(BehaviorLowering.java:191)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:1398)
	at vadl.ast.ViamLowering.visit(ViamLowering.java:108)
	at vadl.ast.PseudoInstructionDefinition.accept(Definition.java:1485)
	at vadl.ast.ViamLowering.fetch(ViamLowering.java:225)
	at java.base@21.0.4/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base@21.0.4/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1708)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
	at java.base@21.0.4/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:622)
	at java.base@21.0.4/java.util.stream.ReferencePipeline.toList(ReferencePipeline.java:627)
	at vadl.ast.ViamLowering.visitIsa(ViamLowering.java:1189)
	at vadl.ast.ViamLowering.findLeafIsa(ViamLowering.java:196)
	at vadl.ast.ViamLowering.generate(ViamLowering.java:153)
	at vadl.cli.BaseCommand.parseToVIAM(BaseCommand.java:241)
	at vadl.cli.BaseCommand.call(BaseCommand.java:305)
	at vadl.cli.BaseCommand.call(BaseCommand.java:62)
	at picocli.CommandLine.executeUserObject(CommandLine.java:2045)
	at picocli.CommandLine.access$1500(CommandLine.java:148)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2465)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2457)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2419)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:2277)
	at picocli.CommandLine$RunLast.execute(CommandLine.java:2421)
	at picocli.CommandLine.execute(CommandLine.java:2174)
	at vadl.cli.Main.main(Main.java:44)
	at java.base@21.0.4/java.lang.invoke.LambdaForm$DMH/sa346b79c.invokeStaticInit(LambdaForm$DMH)
```

flofriday, 2025-05-31:
@AndreasKrall can you help me reproduce this crash?

Currently the code from your snippet is part of the spec on master but checking the code spec doesn't crash the compiler on my end.

AndreasKrall, 2025-06-01:
This code is not on the master. To enable a correct check of `VarRisc.vadl` on the master for simulator generation the indirection of `BRA` via `BGEZ` has been replaced by directly calling `BEQ`. To reproduce the crash you have to replace the `BRA` on the master with the `BRA` in this issue which calls `BEQ` indirectly via `BEQZ`.

flofriday, 2025-06-06:
Ok I could reproduce it.

So the problem at the moment is that I don't quite know yet how the VIAM for that should look like and therefore the lowering throws an exception on purpose.

I will discuss that with @Jozott00 and try to get it implemented soon.

AndreasKrall, 2025-06-07:
I suggest not to change the VIAM, but to do the resolving of the indirection during type checking / lowering as I had done it manually. But I do not know the complexity of this step. If it is too complex for now, currently an error can be thrown and the resolving of the indirection can be moved further in the future after tensor and MiA are finished.

closed 2025-06-07 by Jozott00