OpenVADL/openvadl #382: lcb: Signedness and unsignedness for fields in assembly string

kper, 2025-07-03:
This is a follow up for #380. I am integrating the printing of immediate fields into the assembly printer again and found a potential foot gun which I would like to discuss.

Given this format:

  ```
model UtypeInstr (name : Id, opcode : Bin, rhsEx : Ex) : IsaDefs = {
    instruction $name : Utype =                        // upper immediate instructions
      X(rd) := $rhsEx
    encoding $name = {opcode = $opcode}
    assembly $name = (mnemonic, " ", register(rd), ",", hex(imm))
    }
``` 

We are defining in the assembly direct to print `hex(imm)`. This correspondents to the shifted value for `LUI` and `AUIPC`. The generated C++ printing function will look like that:

```
return "ADDI"s
 + " "s
 + getRegisterNameFromXByIndex(MCOperandWrapper(MI->getOperand(0)).unwrapToIntegral()) // rd
 + ","s
 + getRegisterNameFromXByIndex(MCOperandWrapper(MI->getOperand(1)).unwrapToIntegral()) // rs1
 + ","s
 + AsmUtils::formatImm(RV3264Base_ADDI_imm(MI->getOperand(2).getImm()), 10, &MAI);
```

The interesting part is the following expression `RV3264Base_ADDI_imm(MI->getOperand(2).getImm()`. This expression extracts the operand applies the encoding function for the `imm` field.

However, when I run the test suite I get the following immediate printed.

```
(gdb) p MI->getOperand(2).getImm()
$5 = -16
(gdb) p RV3264Base_ADDI_imm(MI->getOperand(2).getImm())
$6 = 4080
```

In other words, the semantic LLVM immediate value is a signed integer. However, all encoding functions will return an unsigned integer because I can't afford to lose a signed bit. This means during printing the default behavior is to print all raw field immediates as unsigned.
I suppose this issue hasn't come up yet because we have explicitly casted it in the previous implementation. However, that might be only correct for RISC-V and not for other architectures.
The question I have is whether that is acceptable / expected behavior and I should update the riscv specification with the necessary casts to make the fields signed?

Proposal:

`assembly $name = (mnemonic, " ", register(rd), ",", hex(imm as Int<32>))`

@AndreasKrall @benedikt-l-huber

AndreasKrall, 2025-07-03:
First off all, you do not loose a sign bit when moving between unsigned and and unsigned integer. You only have a different interpretation of the same bits, e.g. for a 12 bit immediate the same bits either represent values in the range 0 to 4095 or -2048 to 2047. The only important thing is to know if the assembler expects a signed or unsigned value. So the internal representation is always the same, but it is important to have the correct external representation. If I remember correctly our `hex` is an unsigned external representation and `decimal` is a signed representation. Doing a cast in the argument to `hex` does not change the external representation, it just can extend the number of bits. I am not sure if in all places the correct function, `hex` or `decimal` is used in the VADL specification of RISC-V. Please can you check that with the reference manual.

Of course we also could change the semantics of `hex` and `decimal` to be overloaded for SInt and UInt and give different external representations. But this would be a big change in the type checker.

kper, 2025-07-03:
> First off all, you do not loose a sign bit when moving between unsigned and and unsigned integer. You only have a different interpretation of the same bits, e.g. for a 12 bit immediate the same bits either represent values in the range 0 to 4095 or -2048 to 2047. 

mhmm yes that's true. I didn't think of that way because I thought that it makes sense to keep it always unsigned which requires an extra bit. But of course, that doesn't make much sense.

> If I remember correctly our hex is an unsigned external representation and decimal is a signed representation. Doing a cast in the argument to hex does not change the external representation, it [...]

I couldn't find anything regarding that in the reference manual. Maybe I overread that. But these semantics make sense to me and I am going to implement it that way.

AndreasKrall, 2025-07-03:
> I had to adapt the spec to fix the shift instructions.

That is fine, but we should discuss in the meeting possibilities to increase the capabilities of the assembly builtins.

PR #383 (2025-07-04): lcb: Support raw field immediates in assembly string
This PR adds support for printing raw fields in assembly. It was necessary to apply an encoding function to get the binary representation of the immediate.

Closes #382 
Closes #380