OpenVADL/openvadl #516: frontend: Relational alias types OR dynamic tensor indexing required for ppc64 spec

kbschlosser, 2025-09-21:
PPC64 uses a single 32-bit condition register. This register requires dynamic indexing of both individual bits and 4-bit fields. There are two ways to implement this. 

The first is with relational alias types:
```
register CR  : Bits<32>                       // needed for comparison with upstream
alias register CRF : Bits<3>  -> Bits<4> = CR // field access
alias register CRB : Bits<32> -> Bits<1> = CR // bit access
```
The second is with dynamic tensor indexing:
```
register CR  : Bits<32>              // bit access
alias register CRF : Bits<8><4> = CR // field access
```
Relational alias types support dynamic indexing, but don't support the aliases as written above. Option 2 doesn't work because tensor indexing must be a constant value.

Example with dynamic tensor indexing:
```
instruction set architecture ARCH = {

  program counter PC  : Bits<64>
  memory          MEM : Bits<64> -> Bits<8>

  register CR  : Bits<32>              // bit access
  alias register CRF : Bits<8><4> = CR // field access

  format TestForm : Bits<32> =
  { opcd  : Bits<24>
  , index_bit   : Bits<5>
  , index_field : Bits<3>
  }

  instruction instr : TestForm =
    CRF(index_field) := 0xF               // or CR(index_bit) := 0b1
  encoding instr = {opcd = 1 as Bits<24>}
  assembly instr = ("testinstr")

}

processor EXAMPLE implements ARCH = {
  reset = PC := 0x0000'0000'0000'0000

  [ firmware ]
  [ base: 0x00000000 ]
  memory region [RAM] DRAM in MEM
}
```
causes this error:
```
error: Invalid constant value
     ╭──[file:///_/openvadl/sys/ppc64/issue.vadl:16:9]
     │                                                                                                                                                                               
  16 │     CRF(index_field) := 0xF               // or CR(index_bit) := 0b1                                                                                                          
     │         ^^^^^^^^^^^ Cannot evaluate identifier with origin of vadl.ast.TypedFormatField
     │ 
     Tensor indexing must be a constant value.
```

Either option works, but it’s unclear to me which is easier to implement.

kbschlosser, 2025-09-22:
I think it was probably implemented to represent the general‑purpose registers of AArch. For an instruction like mov reg, #imm, you need a way to dynamically index the register file so that you can do X(reg) := imm.

kbschlosser, 2025-11-07:
Still fails, but with a different error.
```
vadl.viam.graph.ViamGraphError: The resource takes 0 indices but write provided 1
        with node:      (1) ReadRegTensor<Bits<4>, register CR: Bits<32>, null>
        with graph:     Graph{ name='ARCH::CRF Read Behavior', sourceLocation=/_/openvadl/sys/ppc64/issue.vadl:7:3..7:38}
        with graph:     Graph{ name='ARCH::CRF Read Behavior', sourceLocation=/_/openvadl/sys/ppc64/issue.vadl:7:3..7:38}
        with location:  unknown:0..0
```

flofriday, 2025-11-15:
**Note:** Updated minimal example:
```
instruction set architecture ARCH = {
  register CR  : Bits<32>
  alias register CRF : Bits<8><4> = CR

  format TestForm : Bits<24> = { opcd  : Bits<24> }

  instruction instr : TestForm = CRF(0) := 0xF

  encoding instr = {opcd = 1 as Bits<24>}
  assembly instr = ("testinstr")
}
```

Jozott00, 2025-11-20:
Seems like an other issue, please open a new issue with the full stack trace.